vector76/cnc_gcodeviewer #1: use replaces for view model to avoid having to disable native tab

jneilliii, 2022-03-03:
hey @vector76 while reviewing the plugin I noticed you had a note about disabling the native gcode viewer plugin. you might consider using the replaces option of the [get_template_configs](https://docs.octoprint.org/en/master/plugins/mixins.html#templateplugin) callback.

vector76, 2022-03-03:
Ok, I don't know how to do that but I will research it and see what I find.

vector76, 2022-03-03:
Alright, it was not trivial (for me) because I had name collisions (since I had just cloned the existing viewer), but I think this "replaces-viewmodel" approach is now implemented.

Thanks for the feedback, this is nicer, skipping the extra step of disabling the native viewer.

closed 2022-03-03 by vector76