vector76/cnc_gcodeviewer #8: No Plot

JocB1, 2022-04-07:
Hi, 
I installed yesterday the current version of the plugin additionally to the "better grbl support" Plugin. 

Unfortunately, I do not get a plot displayed. 

This is a simply example for a Gcode which I create with Laserweb: 

"
G21         ; Set units to mm
G90         ; Absolute positioning
$X           ; unlock

;
; Operation:    0
; Type:         Laser Cut
; Paths:        1
; Passes:       1
; Cut rate:     200 mm/min
;

; Pass 0

; Pass 0 Path 0
G0 X51.84 Y51.84
M03
G1 X2.00 Y51.84 S30.12 F200
G1 X2.00 Y2.00
G1 X51.84 Y2.00
G1 X51.84 Y51.84
G1 X51.84 Y51.84
M05
M5          ; Switch tool offEnd

"

Can you help me in this case? 

Best regards, 
Jochen

vector76, 2022-04-08:
Thank you for the bug report.  I will see if I can fix this.

vector76, 2022-04-08:
Okay, I think I have a fix for this.  Try 1.0.3 
https://github.com/vector76/cnc_gcodeviewer/archive/refs/tags/v1.0.3.zip

You will need to disable the default built-in gcode viewer for this to work.

I think this will solve your issue.  If it does, then you may close the issue.

JocB1, 2022-04-08:
Hi Jamie,

 

great, thank you
now it runs great!

 

Jochen

 

Gesendet: Freitag, 08. April 2022 um 07:47 Uhr

 

https://github.com/vector76/cnc_gcodeviewer/archive/refs/tags/v1.0.3.zip

closed 2022-04-08 by vector76