vector76/cnc_gcodeviewer #9: No longer works in 1,8

thomasdg, 2022-06-11:


vector76, 2022-06-11:
Thanks for reporting.  I have a spare Pi, so I will try fresh install of 1.8.1 and then the plugin and report back.

thomasdg, 2022-06-24:
Also noticed that the original GCode Viewer plugin is enabled after every reboot... Maybe that has something to do with it?

thomasdg, 2022-06-27:
Not sure if this can give you a clue, but this error appears in the JS console:
Uncaught ReferenceError: AVLTree is not defined
    at prepareLinesIndex (packed_plugins.js?11658008:2981:29)
    at Object.passDataToRenderer (packed_plugins.js?11658008:3085:21)
    at Worker.processMessage (packed_plugins.js?11658008:2744:61)

This is within the GCODE.gCodeReader function definition.

hamanjam, 2022-09-25:
I just came across this last night.  Sadly I'm running 1.8.3 so I"m getting nothing.  I'm running Octoprint on a desktop with Ubuntu so It has a bit more oomph if you need troubleshooting logs

hamanjam, 2022-09-27:
I hacked around and "fixed" the display.  with 1.8x, they updated the viewer code to use double binary instead of AVL tree (as noted above)

I looked through the release notes back then and one note was to remove the file avstree.js from the code.  I went to an older backup and found the file and copied it to my new CNC install.  I copied the file to the equivalent location and edited the init files to load.  Restarted Octoprint and I'm able to see in the gcode viewer

/home/pi/OctoPrint/venv/lib/python3.10/site-packages/octoprint/server/ __init__.py  

------snip-----------------
      js_libs = [
            "js/lib/babel-polyfill.min.js",
            "js/lib/jquery/jquery.min.js",
            "js/lib/modernizr.custom.js",
            "js/lib/lodash.min.js",
            "js/lib/sprintf.min.js",
            "js/lib/knockout.js",
            "js/lib/knockout.mapping-latest.js",
            "js/lib/babel.js",
            **"js/lib/avltree.js",**
            "js/lib/bootstrap/bootstrap.js",
            "js/lib/bootstrap/bootstrap-modalmanager.js",
            "js/lib/bootstrap/bootstrap-modal.js",
            "js/lib/bootstrap/bootstrap-slider.js",
            "js/lib/bootstrap/bootstrap-tabdrop.js",                                                 

---snip---------

 cp /root/avltree.js /home/pi/OctoPrint/venv/lib/python3.10/site-packages/octoprint/static/js/lib/
chown pi.pi /home/pi/OctoPrint/venv/lib/python3.10/site-packages/octoprint/static/js/lib/avltree.js

vector76, 2022-09-28:
Thank you!  I will definitely take a look.

guysoft, 2022-09-30:
Hey, on Octoprint 1.8.4. the avltree.js fix above does not seem to work for me.
However it might be that I am using a plotter and the z movements confuse the plugin. (drawing is a layer below and movement is a layer above)

vector76, 2022-10-01:
Ok I think this is fixed.  I reproduced the issue by upgrading and saw the missing AVLTree error in the js console.  By including avltree.js in the plugin itself, the problem fixed itself and my test code just worked like magic.  Couldn't have been easier, so thanks again @hamanjam !

I'm closing this issue but I am interested to know if there is an issue with 2D jobs (e.g. laser).  @guysoft if the latest release doesn't resolve the issue, please open another issue and I'll be glad to look at a sample.

closed 2022-10-01 by vector76

guysoft, 2022-10-01:
Confirm, this also fixes the issue for me :)

thomasdg, 2022-10-17:
Same here: problem solved. Thanks a lot!